tuupola/micropython-mpu9250 #12: gy91 mpu9255 and new ak8963

hortovanyi, 2019-02-16:
Using a gy-91 which contains a mpu9255 and a different model AK8963

Changed in AK8963.__init__ 
`if 0x48 != self.whoami:`
to
`if self.whoami not in (0x48,0x5F):`

in AK8963.magnetic (was getting the same y an z values, appears you can no longer read six bytes at a time but must read 2 bytes for each of hxl, hyl, hzl)
So changed
`xyz = list(self._register_three_shorts(_HXL)`
to
```
x = self._register_short(_HXL)
y = self._register_short(_HYL)
z = self._register_short(_HZL)
xyz=[x,y,z]
```

and in mpu6500.__init__
`if 0x71 != self.whoami`
to
`if self.whoami not in (0x71,0x73)`

Still working away trying to get this working ...

tuupola, 2019-02-17:
You mean [this](https://sea.banggood.com/GY-91-MPU9250-BMP280-10DOF-Acceleration-Gyroscope-Compass-Nine-Shaft-Sensor-Module-p-1129541.html?cur_warehouse=CN)? Looks quite interesting board. Let me know how it goes.

hortovanyi, 2019-02-18:
Cant seem to get this to work properly .. when I plot all the graphs for calibration the data doesnt look correct. Heres the [MPU-9255 register map](https://stanford.edu/class/ee267/misc/MPU-9255-Register-Map.pdf)

Sample output and the code follows
```
6.53933,-1.681542,-6.165654
-42.7859,-6.165654,-12.14447
-42.41222,-12.14447,-12.14447
-6.165654,-12.14447,-12.14447
-6.165654,-24.84945,-12.14447
-3.176246,-36.05973,-12.14447
-1.681542,-39.04914,-12.14447
6.913006,-41.2912,-15.13388
-37.55444,-39.04914,-12.14447
-30.82827,-24.47578,-12.14447
-36.43341,-6.165654,-12.14447
-37.74128,-5.231464,-15.13388
-42.41222,-46.89634,-12.14447
-42.7859,-46.89634,-12.14447
-0.560514,-42.22539,-12.14447
0.0,-36.43341,-12.33131
-45.77531,-30.08092,-12.14447
3.923598,17.74961,-12.14447
-42.7859,-25.59681,-6.165654
-42.7859,-27.09151,-6.165654
-42.22539,-27.09151,-6.165654
-43.90693,-30.26776,-6.165654
-43.53325,-30.08092,-6.165654
-1.681542,-30.08092,-6.165654
-3.176246,-30.82827,-6.165654
-3.176246,-30.08092,-6.165654
-3.176246,-34.9387,-6.165654
-1.681542,-30.08092,-6.165654
-3.176246,-30.08092,-6.165654
-3.176246,-36.43341,-6.165654
-6.165654,-37.55444,-6.165654
-6.165654,-40.73068,-6.165654
-6.165654,-42.7859,-6.165654
-6.165654,-0.186838,-12.14447
-6.165654,-1.681542,-6.165654
-6.165654,-6.165654,-6.165654
-12.14447,-6.165654,-6.165654
-12.14447,-12.89182,-6.165654
-12.14447,-12.14447,-6.165654
-6.165654,-12.14447,-12.14447
-6.165654,-12.14447,-12.14447
-6.165654,-24.1021,-12.14447
-3.176246,-30.08092,-12.14447
-3.176246,-30.08092,-12.14447
-0.186838,-30.08092,-12.14447
-1.681542,-30.08092,-12.14447
-45.40164,-39.04914,-12.14447
-43.72009,-39.04914,-12.14447
-37.55444,-43.53325,-12.14447
-37.55444,-45.96215,-6.165654
-37.55444,-43.90693,-12.14447
-30.08092,-42.22539,-12.14447
-30.08092,-0.560514,-12.14447
-30.82827,-45.40164,-12.14447
-27.09151,-0.93419,-6.165654
-12.14447,-0.93419,-6.165654
-12.14447,-0.93419,-6.165654
-12.14447,-0.93419,-12.14447
-12.14447,-1.681542,-12.14447
-12.14447,-0.560514,-12.14447
-6.165654,-46.52266,-12.14447
-3.176246,-46.52266,-6.165654
-1.681542,-42.7859,-12.14447
50.81994,-39.79649,-12.14447
53.062,-39.04914,-12.14447
55.11721,-33.81768,-202.7192
[... 242 lines cut ...]
        if value is None:
            self.i2c.readfrom_mem_into(self.address, register, buf)
            return ustruct.unpack("<h", buf)[0]

        ustruct.pack_into("<h", buf, 0, value)
        return self.i2c.writeto_mem(self.address, register, buf)

    def _register_three_shorts(self, register, buf=bytearray(6)):
        self.i2c.readfrom_mem_into(self.address, register, buf)
        return ustruct.unpack("<hhh", buf)

    def _register_char(self, register, value=None, buf=bytearray(1)):
        if value is None:
            self.i2c.readfrom_mem_into(self.address, register, buf)
            return buf[0]

        ustruct.pack_into("<b", buf, 0, value)
        return self.i2c.writeto_mem(self.address, register, buf)

    def __enter__(self):
        return self

    def __exit__(self, exception_type, exception_value, traceback):
        pass 
```

hortovanyi, 2019-02-21:
Getting better data using ` i2c = I2C(scl=Pin(21), sda=Pin(22), freq=100000)`. There appears to be an issue with the default freq on an ESP32 communicating with the GY-91. Might be a timing issues?

Tokelu, 2020-02-26:
@hortovanyi Did you get this to work?
I have been looking into using th GY-91 with a LoPy4.
The LoPy is running µ-Python and getting the IMU to work has but me in some deep waters i have to admit, so I was looking for something to a library or driver for the GY-91

gretel, 2020-11-25:
seems like a common issue.. https://hackaday.io/project/28491-quick-draw-motion-tracking/log/71828-a-spark-a-doh-a-surprise-and-data-sheets

gretel, 2020-11-25:
working fine for me using `MicroPython v1.13-186-g5a7027915 on 2020-11-25; ESP module with ESP8266`:
  * code changed as @hortovanyi posted here
  * `freq=100000`.

ℹ️ the `gy-91` datasheet states 100 and 400khz as the only rates supported.

some seconds running:

```
acc (1.5275, -0.462081, 10.6111)
gyr (0.277387, 3.5177, -0.50388)
mag (10.4332, 20.6402, 1.39219)
tmp 31.0009
--
acc (-0.390255, 0.0766144, 9.69173)
gyr (0.0141225, 0.00106585, -0.0150551)
mag (12.232, 27.8824, 2.78437)
tmp 30.7134
--
acc (-0.301669, 0.0861912, 9.66539)
gyr (0.0222496, -0.0226493, -0.0159877)
mag (12.4119, 27.3392, 1.21816)
tmp 30.5217
--
acc (-0.332794, 0.100556, 9.742)
gyr (0.0209173, -0.0434334, -0.0190521)
mag (11.8723, 27.1582, 2.43633)
tmp 30.5217
--
acc (-0.28491, 0.0742202, 9.61511)
gyr (0.0182527, -0.013856, -0.0123905)
mag (13.3113, 26.434, 3.13242)
tmp 30.6654
--
acc (-0.299275, 0.083797, 9.72046)
gyr (0.0199847, -0.0110582, -0.0163874)
mag (13.3113, 26.0719, 3.13242)
tmp 30.5217
--
acc (-0.299275, 0.047884, 9.64145)
gyr (0.0225161, 0.0214502, -0.0151884)
mag (12.232, 26.434, 2.08828)
tmp 30.0904
--
```

Alex-CodeLab, 2022-01-01:
Any luck with this? 

I'm getting similar results, but it doesn't seem right (only the temp seems to be correct)
```
acce  (-0.02154782, -0.02154782, 9.289502)
gyro  (-0.05249311, 0.01985146, -0.02264931)
magn  (-12.14364, -9.666079, 7.323508)
temp  27.6463

acce  (-0.01197101, -0.06943185, 9.315839)
gyro  (-0.05342574, 0.02251608, -0.01891884)
magn  (-11.57441, -9.490331, 6.794097)
temp  27.79007

acce  (-0.01197101, -0.05027823, 9.344569)
gyro  (-0.05449158, 0.01998469, -0.02118377)
magn  (-11.76415, -10.72056, 5.911747)
temp  27.6463

acce  (-0.01197101, -0.02394202, 9.358934)
gyro  (-0.05315928, 0.02451455, -0.02251608)
magn  (-11.19492, -9.490331, 6.794097)
temp  27.79007

```

ttyyzz34, 2023-04-01:
> Cant seem to get this to work properly .. when I plot all the graphs for calibration the data doesnt look correct. Heres the [MPU-9255 register map](https://stanford.edu/class/ee267/misc/MPU-9255-Register-Map.pdf)
> 
> Sample output and the code follows
> 
> ```
> 6.53933,-1.681542,-6.165654
> -42.7859,-6.165654,-12.14447
> -42.41222,-12.14447,-12.14447
> -6.165654,-12.14447,-12.14447
> -6.165654,-24.84945,-12.14447
> -3.176246,-36.05973,-12.14447
> -1.681542,-39.04914,-12.14447
> 6.913006,-41.2912,-15.13388
> -37.55444,-39.04914,-12.14447
> -30.82827,-24.47578,-12.14447
> -36.43341,-6.165654,-12.14447
> -37.74128,-5.231464,-15.13388
> -42.41222,-46.89634,-12.14447
> -42.7859,-46.89634,-12.14447
> -0.560514,-42.22539,-12.14447
> 0.0,-36.43341,-12.33131
> -45.77531,-30.08092,-12.14447
> 3.923598,17.74961,-12.14447
> -42.7859,-25.59681,-6.165654
> -42.7859,-27.09151,-6.165654
> -42.22539,-27.09151,-6.165654
> -43.90693,-30.26776,-6.165654
> -43.53325,-30.08092,-6.165654
> -1.681542,-30.08092,-6.165654
> -3.176246,-30.82827,-6.165654
> -3.176246,-30.08092,-6.165654
> -3.176246,-34.9387,-6.165654
> -1.681542,-30.08092,-6.165654
> -3.176246,-30.08092,-6.165654
> -3.176246,-36.43341,-6.165654
> -6.165654,-37.55444,-6.165654
> -6.165654,-40.73068,-6.165654
> -6.165654,-42.7859,-6.165654
> -6.165654,-0.186838,-12.14447
> -6.165654,-1.681542,-6.165654
> -6.165654,-6.165654,-6.165654
> -12.14447,-6.165654,-6.165654
> -12.14447,-12.89182,-6.165654
> -12.14447,-12.14447,-6.165654
> -6.165654,-12.14447,-12.14447
> -6.165654,-12.14447,-12.14447
> -6.165654,-24.1021,-12.14447
> -3.176246,-30.08092,-12.14447
> -3.176246,-30.08092,-12.14447
> -0.186838,-30.08092,-12.14447
> -1.681542,-30.08092,-12.14447
> -45.40164,-39.04914,-12.14447
> -43.72009,-39.04914,-12.14447
> -37.55444,-43.53325,-12.14447
> -37.55444,-45.96215,-6.165654
> -37.55444,-43.90693,-12.14447
> -30.08092,-42.22539,-12.14447
> -30.08092,-0.560514,-12.14447
> -30.82827,-45.40164,-12.14447
> -27.09151,-0.93419,-6.165654
> -12.14447,-0.93419,-6.165654
> -12.14447,-0.93419,-6.165654
> -12.14447,-0.93419,-12.14447
> -12.14447,-1.681542,-12.14447
> -12.14447,-0.560514,-12.14447
> -6.165654,-46.52266,-12.14447
> -3.176246,-46.52266,-6.165654
> -1.681542,-42.7859,-12.14447
> 50.81994,-39.79649,-12.14447
> 53.062,-39.04914,-12.14447
[... 250 lines cut ...]
>             return ustruct.unpack("<h", buf)[0]
> 
>         ustruct.pack_into("<h", buf, 0, value)
>         return self.i2c.writeto_mem(self.address, register, buf)
> 
>     def _register_three_shorts(self, register, buf=bytearray(6)):
>         self.i2c.readfrom_mem_into(self.address, register, buf)
>         return ustruct.unpack("<hhh", buf)
> 
>     def _register_char(self, register, value=None, buf=bytearray(1)):
>         if value is None:
>             self.i2c.readfrom_mem_into(self.address, register, buf)
>             return buf[0]
> 
>         ustruct.pack_into("<b", buf, 0, value)
>         return self.i2c.writeto_mem(self.address, register, buf)
> 
>     def __enter__(self):
>         return self
> 
>     def __exit__(self, exception_type, exception_value, traceback):
>         pass
> ```

Where did you change?I can use this code!