tuupola/micropython-mpu9250 #13: [Errno 19] ENODEV

pidou46, 2019-10-07:
Hello,

I'm running into trouble trying "the simple test" code

My mcu :
`MicroPython v1.11-312-g22099ab88 on 2019-09-15; ESP32 module (spiram) with ESP32`

Error:
`import test_mpu
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test_mpu.py", line 7, in <module>
  File "mpu6500.py", line 97, in __init__
  File "mpu6500.py", line 139, in whoami
  File "mpu6500.py", line 155, in _register_char
OSError: [Errno 19] ENODEV`

pidou46, 2019-10-07:
When trying to scan i2c bus I don't see anything

`>>> from machine import Pin,I2C
>>> i2c=I2C(scl=Pin(22),sda=Pin(21))
>>> i2c.scan()
[]
>>>`

I have checked with a VL53L0X and I see it when I scan the bus:

`>>> from machine import Pin,I2C
>>> i2c=I2C(scl=Pin(22),sda=Pin(21))
>>> i2c.scan()
[41]`

It sound like I have a bad hardware...

IHOXOHI, 2020-04-25:
I have the same error (OSError: [Errno 19] ENODEV
) with a pyboard and lcd160, both from original manufacturing... For me, it looks like a problem with the library...

tuupola, 2020-04-25:
Library works just fine. Make sure you have correct pins. If `i2c.scan()` does not find anything it means there is no properly connected i2c devices on those pins.

jasoncook-io, 2020-04-28:
Honestly I think Tuupola has got much more interesting things to deal with in this project (like working on the gyro calibration). @pidou64 take a closer look at the error message. As long as you are getting a result from the i2c.scan() then you can move on to the next step. Make sure that the correct address for your device is stated in the file mpu6500.py, line 88. The hexadecimal by default is 0x68. if you need to get the hexadecimal address for your sensor then try a code like the one [here](https://gist.github.com/projetsdiy/f4330be62589ab9b3da1a4eacc6b6b1c)

IHOXOHI, 2020-04-30:
Hi,
Thanks for your help.
I have found the origin of the problem which come from an other connection of dattalogging on a serial port. No problem without this connection.
Sorry for alls the desagrements.
This Library works fine with pyboard, arduino, ... I enjoy it.
All the best,
Have a fun.

elingeniero3, 2020-10-03:
Hi mates,
I tried the example test code:
```
import utime
from machine import I2C, Pin
from mpu9250 import MPU9250

i2c = I2C(scl=Pin(22), sda=Pin(21))
sensor = MPU9250(i2c)

print("MPU9250 id: " + hex(sensor.whoami))

while True:
    print(sensor.acceleration)
    print(sensor.gyro)
    print(sensor.magnetic)
    print(sensor.temperature)

    utime.sleep_ms(1000)
```
 but I have the sames issues:

`>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "mpu9250.py", line 45, in __init__
  File "mpu6500.py", line 97, in __init__
  File "mpu6500.py", line 149, in whoami
  File "mpu6500.py", line 181, in _register_char
OSError: [Errno 19] ENODEV`

My mcu is:

MicroPython v1.13 on 2020-09-02; ESP32 module with ESP32

I cheked that the pins are correctly connected.
Someone could help me please?

elingeniero3, 2020-10-09:
I've just solved the issue, I didn't correct right the device to the sensor

closed 2020-10-09 by tuupola

APMonitor, 2022-03-31:
The pyboard `PYBv1.1` internal accelerometer creates a conflict with the mpu9250. With `pyb_Acc=True`, there is the OSError 19: ENODEV. When `pyb_Acc=False`, there is no error.

```
[12, 76, 104]
Traceback (most recent call last):
  File "<stdin>", line 15, in <module>
  File "mpu9250.py", line 94, in whoami
  File "mpu6500.py", line 149, in whoami
  File "mpu6500.py", line 181, in _register_char
OSError: [Errno 19] ENODEV
```

Here is the code:

```python
import utime
from machine import SoftI2C, Pin
from mpu9250 import MPU9250
import pyb

pyb_Acc = False

i2c = SoftI2C(scl=Pin('X9'), sda=Pin('X10'))
print(i2c.scan())
s = MPU9250(i2c)

print("MPU9250 id: " + hex(s.whoami))

if pyb_Acc:
    a = pyb.Accel()

while True:
    t = pyb.millis()  # get time
    print(s.acceleration)
    print(s.gyro)
    print(s.magnetic)
    print(s.temperature)
    if pyb_Acc:
        x, y, z = a.filtered_xyz()  # get acceleration data

    utime.sleep_ms(10)
```

I would like to be able to read both the MPU9250 and Internal accelerometer sensors. Any ideas on a way to avoid the conflict in address?

tuupola, 2022-03-31:
How do you know there is a conflict? Does the internal accelerometer use the same address as the attachhed MPU9250?

APMonitor, 2022-03-31:
The `i2c.scan()` command gives `[12, 76, 104]` for pins `X9` and `X10`. It doesn't appear to be a conflict with the I2C address. When`pyb_Acc = True` in the above script, there is an error that the MPU9250 is not found with code `ENODEV`.

Line 181 is from the line `self.i2c.readfrom_mem_into(self.address, register, buf)` of `mpu6500.py`

```python
    def _register_char(self, register, value=None, buf=bytearray(1)):
        if value is None:
            self.i2c.readfrom_mem_into(self.address, register, buf)
            return buf[0]
```

That is called from line 149 of `mpu6500.py`.

```python
    @property
    def whoami(self):
        """ Value of the whoami register. """
        return self._register_char(_WHO_AM_I)
```

The _WHO_AM_I constant is 0x75 (hex) = 117 (decimal). I'm not sure why this register read is having trouble when the pyBoard accelerometer is also reading.

The [micropython documentation](https://github.com/micropython/micropython/blob/master/docs/library/pyb.Accel.rst) for `pyb.Accel()` has a hardware note:

> The accelerometer uses I2C bus 1 to communicate with the processor. Consequently when readings are being taken pins X9 and X10 should be unused (other than for I2C). Other devices using those pins, and which therefore cannot be used concurrently, are UART 1 and Timer 4 channels 1 and 2.

The MPU9250 is I2C so this shouldn't create a conflict unless both accelerometers have the same address. I'll keep digging into it to see if there is something in the `pyb.Accel()` function that is creating the conflict.

tuupola, 2022-04-01:
Ok so `i2c.scan()` can see both MPU6500 `(0x68)` and AK8963 `(0x0c)`. The internal MMA7660 is the `0x4c`. However communicating with MPU6500 fails with `ENODEV`. This error means the device does not respond. You would get the same error if you remove the MPU9250 from the circuit.

Is the `i2c.scan()` result the same with both `pyb_Acc = True` and `pyb_Acc = False`?

APMonitor, 2022-04-02:
The `i2c.scan()` is the same with both `pyb_Acc` (or `mma7660`) as `True` and `False`. The only error is when `mma7660` and `mpu9250` are both `True`.

```python
import utime
from machine import SoftI2C, Pin
from mpu9250 import MPU9250
import pyb

mma7660 = True
mpu9250 = True

i2c = SoftI2C(scl=Pin('X9'), sda=Pin('X10'))
print(i2c.scan())

if mpu9250:
    s = MPU9250(i2c)
    print("MPU9250 id: " + hex(s.whoami))

if mma7660:
    a = pyb.Accel()

while True:
    t = pyb.millis()  # get time
    if mpu9250:
        print(s.acceleration)
        print(s.gyro)
        print(s.magnetic)
        print(s.temperature)
    if mma7660:
        x, y, z = a.filtered_xyz()  # get acceleration data
        print(x,y,z)
    utime.sleep_ms(10)
```

Here are the `i2c` addresses:

- AK8963 (0x0c) 12
- MPU6500 (0x68) 104
- MMA7660 (0x4c) 76

When the MPU9250 is removed, only the (0x4c) 76 address remains.

sriwahyni, 2022-04-20:
>>> i2c.scan()
[104]

why?