tuupola/micropython-mpu9250 #15: RuntimeError: MPU6500 not found in I2C bus.

varna9000, 2020-03-17:
Hi, I've wired the sensor to nodemcu esp8266, however the library si giving me the following error:

`Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
  File "mpu9250.py", line 39, in __init__
  File "mpu6500.py", line 103, in __init__
RuntimeError: MPU6500 not found in I2C bus.`

Wiring: CLK->pin D1, SDA->pin D2.

i2c scanner shows:

> Scan i2c bus...
i2c devices found: 1
Decimal address:  104  | Hex address:  0x68

What could be wrong?

tuupola, 2020-03-17:
You could try to debug what whoami register contains.

https://github.com/tuupola/micropython-mpu9250/blob/master/mpu6500.py#L102

varna9000, 2020-03-17:
Can you give me a clue how can I do that? I'm not that much into registers :)
 The board I'm using is this: 

https://www.tinytronics.nl/shop/image/cache/data/product-493/mpu-9250-1000x1000.jpg

I tried to find a Datasheet for this particular board and find the whoami register value, but there is no definite information anywhere. Is there a way to query the board and check it?

varna9000, 2020-03-17:
Ok, got some progress. I printed self.whoami and the hex which came out is 0x70. So this works fine now when I replaced your default 0x71. Now same problem is with ak8963.py:

> Traceback (most recent call last):
>   File "<stdin>", line 10, in <module>
>   File "ak8963.py", line 76, in __init__
>   File "ak8963.py", line 140, in whoami
>   File "ak8963.py", line 198, in _register_char
> OSError: [Errno 19] ENODEV

Will try same approach and report back here.

varna9000, 2020-03-17:
I think, this board doesn't have a magnetometer.... should i2c scan return ak8963 address as well?

tuupola, 2020-03-18:
Quickly Googling the MPU-92/65 breakout board you have, should have the MPU-9250. MPU-9250 is a System in Package (SiP) which combines two chips: MPU-6500 and an AK8963.

The value of MPU-9250 `WHO_AM_I` register is `0x71`. The value of MPU-6500 `WHO_AM_I` register is `0x70` and default address is `0x68`. So if I am reading my code correctly you just found a bug. I seem to have a wrong value. I don't really understand why I have not caught this before.

To directly access the magnetometer it has to be in bypass mode. Bypass mode is enabled at [mpu6500.py#L112-L115](https://github.com/tuupola/micropython-mpu9250/blob/master/mpu6500.py#L112-L115). AK8963 address should be either `0x0c`, `0x0d`, `0x0e` or `0x0f` depending on the configuration.

So maybe your breakout board only has the MPU-6500. Other option is that the bypass mode is not working for some reason. You could try checking with magnifying glass what the chip on the breakout board is.

varna9000, 2020-03-18:
As far as I can read the chip ID it's MP65 6799D1 L1506
I'll try to find a Datasheet and also I'll check also the addresses you mentioned and see if I can get the magnetometer (if any) to work.
 
[image]

tuupola, 2020-03-22:
MP65 suggests the chip is the standalone MPU-6500.  Also the `WHOAMI` value suggests it is the standalone MPU-6500 and not the MPU-9250 SIP. In other words it does not include the magnetometer.

I have only been testing with MPU-9250 which was why I did not catch earlier that the standalone accelerometer has different `WHOAMI` value. Will commit a fix soon. Thanks for the heads up!

tuupola, 2020-03-22:
You should be able to use the standalone MPU6500 witch something like following:

```python
from machine import I2C, Pin
from mpu6500 import MPU6500

i2c = I2C(scl=Pin(22), sda=Pin(21))
sensor = MPU6500(i2c)

print("MPU6500 id: " + hex(sensor.whoami))
```

closed 2020-03-22 by tuupola

varna9000, 2020-03-22:
Using only MPU6500 class, chip works perfect :) Thanks for the great library!

commit 7dd5ece6 (2020-03-22): Allow using standalone MPU6500 (fixes #15) (#16)

The MPU6500 inside the MPU9250 SIP has different WHOAMI value.
This commit allows using also the standalone MPU6500. 

https://www.cdiweb.com/datasheets/invensense/tdk-invensense-mpu9250-and-mpu6500.pdf