tuupola/micropython-mpu9250 #21: MPU6500 class fails to initialize when used from MPU9250 class

k79fi, 2020-05-03:
1. Code snippet:

from machine import Pin, I2C
from mpu6500 import MPU6500
from mpu9250 import MPU9250

i2c=I2C(scl=Pin(22), sda=Pin(21))
sensor=MPU9250(i2c)
...

2. Expected result: MPU9250 class successfully instantiated

3. Actual result: code throws runtime exception "MPU6500 not found in I2C bus."

4. Suspected code:
        if self.whoami not in [0x71, 0x70]:
            raise RuntimeError("MPU6500 not found in I2C bus.")

MPU9250 has i2c address 0x68, however here it compares it to 0x70 and 0x71 and fails if comparisons fail.
Once this code is commented out, the library seems to work just fine.

tuupola, 2020-05-04:
The line 

```python
if self.whoami not in [0x71, 0x70]:
``` 

does not compare the i2c address. Instead it compares the value of the `WHO_AM_I` register.

For MPU-9250 the value of `WHO_AM_I` register is `0x71` and address i2c is `0x68` or `0x69`. The value of MPU-6500 WHO_AM_I register is `0x70` and address is `0x68` or `0x69`.

Could you check what is the value of `self.whoami`? Maybe you have a custom chip, or chip which is compatible but not actual MPU-9250.

tuupola, 2020-11-25:
No feedback.

closed 2020-11-25 by tuupola

phuzybuny, 2021-07-04:
Encountering the same issue as OP.

Using `print(self.whoami)` the value I got was `115` - I believe this corresponds to `0x73`.

The module I am using is the same as the one described [here](https://components101.com/sensors/MPU9250-9-dof-mems-sensor-module-datasheet-pinout-features-working).