tuupola/micropython-mpu9250 #23: TypeError: 'id' argument required

Robotayylien, 2021-08-19:
When I try to run the simple test I am getting this error:

♦Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
TypeError: 'id' argument required

Robotayylien, 2021-08-19:
when I add id=0 to i2c I get this error

```♦Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
  File "mpu9250.py", line 56, in __init__
  File "ak8963.py", line 76, in __init__
  File "ak8963.py", line 139, in whoami
  File "ak8963.py", line 197, in _register_char
OSError: [Errno 5] EIO```

Robotayylien, 2021-10-24:
Yeah I have it set to id 0 for pin 0 and 1, using the pi pico board, if i did change the id to 1 with the same pins I get a ValueError: bad SCL pin

JoelTech43, 2023-09-06:
@Robotayylien I ran into a similar issue where if I used ```id=1``` I got ```ValueError: bad SCL pin``` and if I put ```id=0``` I got OSError. I was using an lcd screen and found that I needed to use ```id=0``` but I needed to change the I2C address. I had the address for my lcd set to ```0x27``` when I needed it set to ```0x3f``` and changing this solved it.

I found the address using this code:
```from machine import I2C, Pin
i2c = I2C(0, sda=Pin(8), scl=Pin(9), freq=400000)
I2C_ADDR = i2c.scan()[0]
print(hex(I2C_ADDR))```
Obviously you need to change the pin numbers in line 2!

I don't know if this will help!

RuudKapteijn, 2023-12-03:
The id parameter can be ommitted. Solved the issue for me on ESP32, MicroPython, MPU9250.

tuupola, 2023-12-04:
Neither this library nor the examples have code which includes the [id in I2C initialisation](https://github.com/search?q=repo%3Atuupola%2Fmicropython-mpu9250+%22%3D+I2C%22&type=code). Looking at MicroPython [I2C documentation](https://docs.micropython.org/en/latest/library/machine.I2C.html) passing id is optional and the value depends on the board you are using.

RuudKapteijn, 2023-12-04:
[image]

`i2c = I2C(0, sda=Pin(8), scl=Pin(9), freq=400000)`

Given de constructor definition I'd assume the first 0 in the constructor call of Robotayylien above is the id parameter. My suggestion is to omit this zero. It did fix an error in my configuration. Do I miss something?

closed 2024-03-19 by tuupola