tuupola/micropython-mpu9250 #25: OSError: [Errno 5] EIO

BarM8, 2022-10-14:
Hi, I'm running into the following error and I can't figure out for the life of me how to resolve it.

Traceback (most recent call last):
  File "<stdin>", line 25, in <module>
  File "mpu9250.py", line 45, in __init__
  File "mpu6500.py", line 97, in __init__
  File "mpu6500.py", line 149, in whoami
  File "mpu6500.py", line 181, in _register_char
OSError: [Errno 5] EIO

Could anyone shed some light on the underlying problem here? It seems others with similar problems were able to resolve it via re-inspecting and connecting the board and sensors but I'm having no such luck. Thanks.

tuupola, 2022-10-14:
Can you see the devices if you do a `ì2c.scan()`

BarM8, 2022-10-14:
Apologies, I could've specified that I do see a device at dec 104, hexa 0x68.

project2501x, 2022-10-15:
I've resolved this by blanking my main.py script. 

My thinking is that main.py, which contained code pertaining to a different sensor, was being executed upon powering the Pico up by connecting it to my PC / Thonny. The code was initialising an object relating to that other sensor and in the process, possibly assigning a different constant value to the WHOAMI register or to a relevant address. I presently lack the understanding to investigate further but will persist to better grasp the error.

Thank you for your prompt response Tuupola, and this great library.

closed 2022-10-15 by project2501x