tuupola/micropython-mpu9250 #26: AK8963 - [Errno 5] EIO

serdaraltin, 2022-11-11:
[image]
[image]

I am using sensor model MPU9250 + BMP280 and I cannot read the magnetometer sensor.

I can read the data of all other sensors, but the sensor named AK8963 gives the error in the picture.

Thanks

tuupola, 2022-11-12:
The `i2c.scan()` finds `104` ie `0x68` which is default i2c address for MPU-9050 and `118` ie `0x76` which is default i2c address for BME280. I cannot find MPU-9050 datasheet at the moment but it seems it is not compatible with MPU-9250.

serdaraltin, 2022-11-12:
When I turn off data reading from AK8963 sensor other sensors work fine.
Why is this so?
[image]
[image]
[image]

tuupola, 2022-11-18:
There is no AK8963 in the i2c bus. There is only BME280 and MPU-6500. So my guess is that your sensor is not MPU-9250 but a plain MPU-6500 instead.

MPU-9250 is a System in Package (SiP) which combines two chips: MPU-6500 and AK8963.