tuupola/micropython-mpu9250 #27: Can not use AK8963

ttyyzz34, 2023-04-01:
Who can help me to solve the issue?
I can not find AK8963.
I use i2c.scan() get 
i2c.scan()
[60, 104, 118]
Can not find AK8963.
I use the AK8963.py and it give error like this:
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
  File "ak8963.py", line 76, in init
  File "ak8963.py", line 139, in whoami
  File "ak8963.py", line 197, in _register_char
OSError: [Errno 19] ENODEV

jchome, 2025-02-20:
Hello,
I've the same problem : 
`̀`
  File "mpu6500.py", line 157, in whoami
  File "mpu6500.py", line 189, in _register_char
OSError: [Errno 19] ENODEV
```

I've connected the MPU with (I think) correct pins.

tuupola, 2025-02-20:
What does `print(i2c.scan())` return?

jchome, 2025-02-20:
> What does `print(i2c.scan())` return?

It returns an empty array : []

My pins seems to be wrong...